jacroe/pidora #70: Naming problems

iamale, 2015-05-14:
1. There's already a thing called 'Pidora', it's [a Fedora remix for Raspberry Pi](http://pidora.ca/).
2. As have been noted by the-other-Pidora team, [it doesn't sound quite good among Russian-speaking community](http://zenit.senecac.on.ca/wiki/index.php/Pidora_Russian).

I suppose the name should be changed. What do you think?

billchurch, 2015-05-14:
PandorPi? Google search comes back relatively clean, which is good.

jacroe, 2015-05-14:
Well, *technically*, I had the name first ([Oct 2012](https://github.com/jacroe/pidora/commit/a80b7dad571f3461f0827cc2469d612c4b393b3a) vs [May 2013](http://pidora.ca/pidora/releases/18/release-notes.txt)). So they're the ones that should have to change. :wink:

But I'm all for a naming change. PandorPi is pretty good. Are there any other suggestions?

iamale, 2015-05-14:
> Well, technically, I had the name first
Oh :>